RoboSense-LiDAR/rslidar_sdk #174: Distructive Interpolation of Airy lidar point clouds

Marselka, 2025-05-06:
The problem is that lidar point clouds are interpolated when should not.

Take a look at screenshots of visualization of some pcd extracted from ROS2 bag file. 
These are pcds of the office floor with many round lamps that are hanging on three thin strings each.

There these smooth lines underlined by rectangles should not exist since there is nothing in these areas.

**Please, let me assure you, colleagues, there is nothing inside these rectangles in reality, but pcds show smooth lines connecting some real but disconnected objects.**

**What is wrong with that lidar or configurations in both GUI and ROS2 SDK?**

[image]
[image]
[image]

We have tested ROS2 driver
- with the setups as is and 
- with changing all the configs in config file https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/main/config/config.yaml

We have tested original GUI from Robosense as well and changed any params there

georgflick, 2025-05-12:
You could try to increase the `Trail Filter Level` via the web interface of the robosense airy (per default on http://192.168.1.200 ).
That should reduce these points, but if it's set too aggressive, it will filter out also some other points.

AmonYang2471, 2025-07-17:
There is a problem of suction and dragging points. The optimization effect of suction and dragging points in the subsequent new firmware will be better.

closed 2025-07-17 by AmonYang2471