RoboSense-LiDAR/rslidar_sdk #176: [Duplicate from rs_lidar issue] No timestamp field in ROS2 PointCloud2 message

Marselka, 2025-05-14:
We use FastLIO-like Lidar-Inertial Odometry algorithm that **needs the timestamp of every point** in the lidar pointcloud.

This requirement is satisfied at least by **Velodyne VLP-16** and **Livox MID360** lidar models. However, the combo of **Robosense Airy** + ROS2 rs_driver cannot provide timestamp field. 
`ros2 topic echo /rlidar_points`:

```
header:
  stamp:
    sec: 668
    nanosec: 403185160
  frame_id: base_scan
height: 900
width: 96
fields:
- name: x
  offset: 0
  datatype: 7
  count: 1
- name: y
  offset: 4
  datatype: 7
  count: 1
- name: z
  offset: 8
  datatype: 7
  count: 1
- name: intensity
  offset: 12
  datatype: 7
  count: 1
is_bigendian: false
point_step: 16
row_step: 1536
data:
- 41
- 169
- 139
...
```

Is it possible to add timestamp field to the PointCloud2 ROS2 message in the driver?

georgflick, 2025-05-14:
You need to change the PointType in the CMakeLists.txt from `XYZI` to something like `XYZIRT`
https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/b897430d98ec9f31ac293fad872607de5071e2b6/CMakeLists.txt#L8

Marselka, 2025-05-14:
@georgflick Thanks a lot!

closed 2025-05-14 by Marselka

PAyush15, 2025-11-28:
> [@georgflick](https://github.com/georgflick) Thanks a lot!

Hey @Marselka, I am also doing something similar. I am also using FAST-LIO2 with the Robosense Airy for SLAM using ROS2. 
Can you please explain a bit how you did it? Like did you directly use the rslidar data for fast lio or did you convert it to velodyne or how is it? I am working on something similar, so would be very helpful.
Thank you very much! Best regards.