RoboSense-LiDAR/rslidar_sdk #177: Setting IP address of E1R

agawel-bdai, 2025-05-14:
Hi all,
Similar to #59 I would like to change the IP address of an E1R sensor, but the documentation and available tools are not helpful in this regard. Is there a separate method available to set IPs?

SotirisCds, 2025-05-14:
I'm also encountering the same issue with the E1R LiDAR. Attempting to access its configuration via the default IP address (192.168.1.200) through a web browser doesn't yield any results. The available documentation lacks guidance on changing the default IP address, and tools like RSView don't appear to support configuration changes. It would be very helpful if RoboSense could provide a dedicated utility or detailed instructions for configuring the E1R's network settings. Thanks!

SotirisCds, 2025-05-15:
**Update:** I reached out to RoboSense support and received a response. They provided a software tool and a user manual that detail how to configure the E1R LiDAR's IP address and other settings. You can access them here: https://robosense.feishu.cn/docx/FCh3dhcLnobspPx5ZqAcTntwnuh

agawel-bdai, 2025-05-19:
Thank you! I received the same response and was able to set the IP.

JamalHsu, 2025-05-22:
Hi I have a M1 lidar that I can't configuration the IP and the error code as below

  931:549 [Debug] Routing active response message.
   937:204 [Debug] "1060"
   937:329 [Warn ] NRC 12
   937:329 [Debug] "2761"
   937:469 [Warn ] NRC 7F
   937:469 [Debug] "2762"
   937:589 [Warn ] NRC 7F
   937:589 [Debug] "2efe81010203040506"
   937:729 [Warn ] NRC 7F
   937:729 [Debug] "1060"
   937:869 [Warn ] NRC 12
   937:869 [Debug] "2761"
   938:008 [Warn ] NRC 7F
   938:009 [Debug] "2762"
   938:149 [Warn ] NRC 7F
   938:149 [Debug] "2efe84ffffff00"
   938:288 [Warn ] NRC 7F
   938:289 [Debug] "1060"
   938:409 [Warn ] NRC 12
   938:409 [Debug] "2761"
   938:549 [Warn ] NRC 7F
   938:549 [Debug] "2762"
   938:689 [Warn ] NRC 7F
   938:689 [Debug] "2efe82c0a801c8"
   938:829 [Warn ] NRC 7F
   938:829 [Debug] "1060"
   938:969 [Warn ] NRC 12
   938:969 [Debug] "2761"
   939:109 [Warn ] NRC 7F
   939:109 [Debug] "2762"
   939:249 [Warn ] NRC 7F
   939:249 [Debug] "2efe86c0a80166"
   939:368 [Warn ] NRC 7F
   939:369 [Debug] "1060"
   939:509 [Warn ] NRC 12
   939:509 [Debug] "2761"
   939:649 [Warn ] NRC 7F
   939:649 [Debug] "2762"
   939:789 [Warn ] NRC 7F
   939:789 [Debug] "2efe831

SchmidL, 2026-10-13:
The newer software to do it is: https://robosense.feishu.cn/wiki/HowCwcIBdiKpimkSvRYc9Vgenbd (or at least it looks newer from the date)