RoboSense-LiDAR/rslidar_sdk #206: E1R与相机进行标定

wzhaofeng, 2025-09-15:
E1R激光雷达与相机联合标定有无推荐的方法

wzhaofeng, 2025-09-26:
> 可以试试这个手动标定方法 https://github.com/pixmoving-moveit/manual_lidar_camera_calibration

你也使用的E1R雷达吗，我是用HKU的fast-calib标定了，想用来跑fast-livo2，但是现在fast-livo2运行又出问题了

ahuazuipiaoliang, 2025-09-26:
> > 可以试试这个手动标定方法 https://github.com/pixmoving-moveit/manual_lidar_camera_calibration
> 
> 你也使用的E1R雷达吗，我是用HKU的fast-calib标定了，想用来跑fast-livo2，但是现在fast-livo2运行又出问题了

Airy, E1R, M1P都用过，密度越高越好，不过是人工选择角点，所以兼容所有激光雷达

wzhaofeng, 2025-09-26:
> > > 可以试试这个手动标定方法 https://github.com/pixmoving-moveit/manual_lidar_camera_calibration
> > 
> > 
> > 你也使用的E1R雷达吗，我是用HKU的fast-calib标定了，想用来跑fast-livo2，但是现在fast-livo2运行又出问题了
> 
> Airy, E1R, M1P都用过，密度越高越好，不过是人工选择角点，所以兼容所有激光雷达

哦哦了解了，感谢回复

OKAIOK, 2025-11-06:
> > 可以试试这个手动标定方法 https://github.com/pixmoving-moveit/manual_lidar_camera_calibration
> 
> 你也使用的E1R雷达吗，我是用HKU的fast-calib标定了，想用来跑fast-livo2，但是现在fast-livo2运行又出问题了

您好，我最近想使用e1r进行雷达与rtk间的标定，请问您做过这方面的工作吗？能否交流一下

wzhaofeng, 2025-11-07:
> > > 可以试试这个手动标定方法 https://github.com/pixmoving-moveit/manual_lidar_camera_calibration
> > 
> > 
> > 你也使用的E1R雷达吗，我是用HKU的fast-calib标定了，想用来跑fast-livo2，但是现在fast-livo2运行又出问题了
> 
> 您好，我最近想使用e1r进行雷达与rtk间的标定，请问您做过这方面的工作吗？能否交流一下

这个我没做过

wzhaofeng, 2025-11-27:
> 可以试试这个手动标定方法 https://github.com/pixmoving-moveit/manual_lidar_camera_calibration

你好，这个标定板文件有吗，还是说自己制作