RoboSense-LiDAR/rslidar_sdk #209: E1R：point_type XYZIRT cann't work

wzhaofeng, 2025-09-28:
I want to use the XYZIRT point type. I modified the point_type in config.yaml using gedit and then recompiled with catkin_make, but it still didn’t work.

[image]

[image]

YiKang-Liu, 2025-10-06:
I've encountered this issue too. Have you managed to resolve it?

YiKang-Liu, 2025-10-06:
[image]
In the CMakeLists.txt file of the rslidar_sdk, there is an option like this. Setting it to XYZIRT here should enable the corresponding data output. This is how I resolved it.